EremusOne/ESPectrum #24: ESPectrum 1.0 rc1 on ESP32-SBC-FabGL SD card issue

Magnetrix, 2023-07-05:
Hello,
When i select external Sd as my storage option, and press F5, no games list is presented.
This card when inserted om my other board TTGo vga32 works great.
Anyone can help me?
Regards

Magnetrix, 2023-07-05:
I am not a programmer, so I don't have skills to know how compile the code. The internal speaker of the board doesn't output sound... sound only outputs from jack 3.5mm... 
[image]

Magnetrix, 2023-07-05:
[ESP32-SBC-FabGL_Rev_A.pdf](https://github.com/EremusOne/ZX-ESPectrum-IDF/files/11960133/ESP32-SBC-FabGL_Rev_A.pdf)

EremusOne, 2023-07-05:
Hi again Magnetrix,

As I told you before, we own no Olimex board to work with and test the emu so we can't test any fix on it. We only release binaries for boards or products we have been able to test thoroughly. Right now, these are the Lilygo TTGO VGA32 and Antonio Villena's ESPectrum board.

This is done this way for a good reason: to reduce this kind of issues we are talking of right now. Bear in mind we work in this project in our spare time and have to be cautious with the compromises we assume.

Anyway, if u have managed to flash binaries on the board I'm sure you'll be able to install VSCode, download the source, apply the suggested fix, compile it and flash to the board. It's easier than you may think.

Good luck and tyvm for your interest in ESPectrum!

closed 2023-07-05 by EremusOne