EremusOne/ESPectrum #25: ESP32-SBC-FabGL open menu issue

Magnetrix, 2023-07-20:
Hello,
With your clues, I can now compiling and installing firmware on board with Visual Code Studio (PlatformIO).

I did a Clone of your repo and modified hardpins.h as you told me on the last post.
So I compiling and upload to the board.
Now, when I power the board, on my VGA monitor I can see spectrum default image align to left.
If I press F1 to access menus, the behavior is like if I keep pressing the "UP" key from keybaord, and could not stop the rolling selected option from the menu. 

Note:
If I flash the board using you bin file from the releases section with ESP32 Flash Download Tool, the menus are working great without that issue, and image is align to center. However can't access SD card as discussed before.

Do you have any suggestion???

Regards

EremusOne, 2023-07-21:
Hi Magnetrix, glad to see you're getting results :)

The problem you've seen is easy to resolve: you've two PlatformIO project environments defined in the platformio.ini file: pico32 and ESPectrum.

You must compile the pico32 environment and all should be working right.

ESPectrum environment is for Antonio Villena's ESPectrum board which has support for real ZX Spectrum kbd ( https://antoniovillena.es/store/product/espectrum ).

Regards ;)

P.S.: In next release there will be no diference: the code will be capable of detect the board is running on.

Magnetrix, 2023-07-21:
yuuuupppiiiiiiiiii

Works like a charm :)
Thank you guys... 
I am supporting your wonderful work via patreon... Congratulations for your fantastic work...
Hope I get access to the source code for rc2 version and modify hardpins.h in the SD section in order to have my ESP-SBC-FabGL updated and fully working

Regards

So in conclusion this board is 100% compatible and only needs to modify hardpins.h, where #define PIN_NUM_MISO GPIO_NUM_2 changes to #define PIN_NUM_MISO GPIO_NUM_35, to access SD card content

Reading 
https://olimex.wordpress.com/2023/05/31/zx-espectrum-emulator-tested-with-esp32-sbc-fabgl-the-mensch-computer-with-w65c256-got-vga-display-and-keyboard/

I think that the issues with menus refered are the same that I described to you before, and simply needs to discard [env:ESPectrum] section in platformIO.ini

Best regards

closed 2023-07-21 by Magnetrix