EremusOne/ESPectrum #36: Libretro core

leiradel, 2024-01-07:
Hi,

The [Libretro](https://www.libretro.com/) project is widely used in the emulation scene for its wide arrays of emulated computers, consoles, and game engines.

The project is centered around the concepts of cores, which are emulators built as dynamic libraries that use the [Libretro API](https://www.libretro.com/index.php/api/), and a frontend, which is an application that interfaces with the OS and is capable of loading a core and rendering its audio and video frames, and directs input events to it. [RetroArch](https://www.retroarch.com/) is the official Libretro frontend, but others exist.

It currently has a [ZX Spectrum core](https://github.com/libretro/fuse-libretro/), ported from the [Fuse emulator](https://fuse-emulator.sourceforge.net/), which I'm the original author, but I'd like to port the ZX-ESPectrum-IDF code to the Libretro API. Would the authors of this code be willing to change the license to LGPL, so it could be used with any Libretro frontend?

Best regards,

Andre

EremusOne, 2024-01-08:
Hi Andre,

Unfortunately, ESPectrum uses parts of code that are under GPL license so we can't change our license to LGPL without specific consent of the authors.

Anyway, thanks for you interest in our project :)

Best regards

leiradel, 2024-01-08:
Oh sorry, I thought the code was all original.

Thanks for answering.

Best regards.

closed 2024-01-08 by leiradel