EremusOne/ESPectrum #37: ESPjoy

Gods69, 2024-01-10:
I own an ESPectrum and an ESPjoy from Antonio Villena. With this version 1.0, I cannot use my joystick. 
I saw that it was necessary to update the firmware of the card. 
On the github of the indicated page, the programming .hex file does not seem to be up to date, so I went through the Arduino IDE. 
With an AVRasp programmer and configuring the target board as "Arduino Leonardo", it seems to me that it worked. 
But, it still doesn't work in the emulator. 
I may be confused with the menu with a PS/2 option and another joystick option.
Someone explain to me how it works?
THANKS.

EremusOne, 2024-01-10:
Hi there,

PS/2 joystick menu isn't related to ESPjoy. It's just for using PS/2 keyboard to simulate joystick input.

Make sure you have the "Keyboard" option selected in "Options/Other/Second PS/2 Device" to tell the emulator to read ESPjoy input from second PS/2 port.

Make sure also that you connect the ESPjoy to the USB connector that is next to the green audio input.

Once you've done this, go to "Joystick/Joystick 1 (or 2 if you connected your joystick to the second DB9 port)/Assign keys" and use the "JoyTest" option to test joystick input.

If arrows and/or buttons go green when you press it on the joystick you're done :)

Good luck ;)

Gods69, 2024-01-10:
It's better now ! Thanks. On the other hand, I can't find the option to assign the Fire 1 button on the Kempston joystick. When I press the fire button on the joystick I have the circle with the letter B which lights up green in the test.

EremusOne, 2024-01-10:
To be able to assign Kempston or Fuller actions you have to select the corresponding type of joystick before in the same menu where you have the "Assign Keys" option.

Gods69, 2024-01-10:
It works now.. Thanks for your time.

closed 2024-01-10 by Gods69