EremusOne/ESPectrum #45: How do you flash a custom ROM?

SanPollo, 2024-04-03:
The README.md says:

`Possibility of using one 48K and one 128K custom ROM with easy flashing procedure from SD card.`

However, I can find no documentation explaining how to do this. I have put a .rom file on the root of the SD card, but whenever I choose **Update** -> **Custom ROM 48K** or **Custom ROM 128 k** -> **Yes**, it says **No custom ROM file found.**

What should the custom ROM files be called? And where should they be placed?

Also, please can this information be added to the README.md so it's clear for anyone else with the same question?

Thanks in advance!

bleughb, 2024-04-03:
Just had a search in the code - 

If you have a look at 

in OSDMAIN.CPP it's looking for the files
/sd/48custom.rom   
and
/sd/128custom.rom

 to bring up that error message

try renaming the files

EremusOne, 2024-04-04:
Hi SanPollo,

bleughb (tyvm!) told you right about the solution.

I told Patreons how to do this but forgot to explain it in Readme.md sorry O:-)

Here are the instructions:

**Instructions for use:**

Two custom ROMs can be installed: one for the 48K architecture and another for the 128K architecture.

The "Update firmware" option is now changed to the "Update" menu with three options: firmware, custom ROM 48K, and custom ROM 128K.

Just like updating the firmware requires a file named "firmware.bin" in the root directory of the SD card, for the emulator to install the custom ROMs, the files must be placed in the mentioned root directory and named as "48custom.rom" and "128custom.rom" respectively.

For the 48K architecture, the ROM file size must be 16384 bytes.

For the 128K architecture, it can be either 16kb or 32kb. If it's 16kb, the second bank of the custom ROM will be flashed with the second bank of the standard Sinclair 128K ROM.

It is important to note that for custom ROMs, fast loading of taps can be used, but the loading should be started manually, considering the possibility that the "traps" of the ROM loading routine might not work depending on the flashed ROM. For example, with Rodolfo Guerra's ROMs, both loading and recording traps using the SAVE command work perfectly.

Finally, keep in mind that when updating the firmware, you will need to re-flash the custom ROMs afterward, so I recommend leaving the files "48custom.rom" and "128custom.rom" on the card for the custom ROMs you wish to use.

Regards ;)

closed 2024-04-05 by EremusOne

SanPollo, 2024-04-05:
Thanks, both. I was hoping to flash the +3e ROM, but it seems it is 64K in total, so I guess it wouldn't be possible at this stage?

EremusOne, 2024-04-09:
Nope. +2A and +3 emulation is on the roadmap but still undone.