EremusOne/ESPectrum #47: Typing quickly does not register all keypresses

SanPollo, 2024-04-15:
I have been trying some non-game programs in the emulator, such as 128K BASIC, HiSoft Pascal Compiler, and Tasword.

For all of them, if you type too quickly, not all of the characters appear on the screen.

Please could this be fixed in a future release?

EremusOne, 2024-04-17:
Hi!

Could you give more context about this matter?

ESPectrum version, hardware used, ¿Did you compare behaviour with some other emulators?, ¿Did you compare also with some real Speccy?

Thanks in advance ;)

SanPollo, 2024-04-17:
I'm running ESPectrum 1.1 on an Olimex ESP32-SBC-FabGL with a Gigabyte GK-K5000 PS/2 keyboard.

I have tested the same keyboard in two PCs, and two other ESP32-SBC-FabGL boards running emulators built with the FabGL library (RunCPM and Multitasking CP/M).

I don't have a real Speccy to test with at the moment. However, I cannot find any known issues on the internet with Spectrums not registering all keystrokes when the user types too quickly.

I'm not sure if this is a problem with ESPectrum itself, or maybe with the underlying ESP32Lib library.

Here are some screenshots to illustrate the issue:

[image]

[image]

EremusOne, 2024-05-06:
Hi again!

Sorry for the delay in my answer. Very busy weeks here!

Right now I don't have my real Speccy to check the behaviour on a real machine but I've asked to some other emulator authors and they all conclude that missing keystrokes on a real Spectrum isn't that strange. I've tested fast typing in some other emulators and they all miss some keystrokes so it doesn't seem a problem of the emulators but the normal behaviour of the real machine.

ESPectrum polls the keyboard 50 times every second so I think it should be enough for most cases. Anyway, if I have the time, in the future I'll try some different approach for keyboard emulation but maybe it will not change anything.

Regards!

SanPollo, 2024-05-06:
Many thanks for your reply. I've got my Spectrum +2A now, so I will test it on there, and let you know the result.

To explain why this is important to me - I would like to do more than just play games on my machine, and I type fairly quickly. I don't type at lightning speed, but "fluently" I guess is the word. I certainly can't type 50 letters a second!

I will post again as soon as I have done my tests.

SanPollo, 2024-05-07:
[image]

I tried typing quickly on my +2A today and all of the keys were registered.

EremusOne, 2024-05-12:
As I told before, in the future, I'll try the mentioned approach so we could test if there's some improvement.

Meantime, there's some roadmap to comply with :D

Thanks again for taking your time for testing this :)

closed 2024-05-12 by EremusOne

SanPollo, 2024-05-12:
Understood, but why close this issue if it the problem hasn't been fixed?

EremusOne, 2024-05-12:
> Understood, but why close this issue if it the problem hasn't been fixed?

Because I don't think of this as a problem to fix. Actual keyboard code is as good as many other emulators keyb code out there.

Anyway, as I said, when I have the time I'll consider your observations and try to find if some changes to keyb code translates into some difference in the scenarios you described. It will need more testing in different models and situations and that means time I don't have right now.

Regards ;)