EremusOne/ESPectrum #48: PS/2 to Joy shematics

soviet9922, 2024-04-24:
Hi looked in the code and documentation but can't find a schematic on how to connect the db9 joystick ports to the ps/2 mouse port.

Where i can find it ?

Gods69, 2024-04-25:
Hi, you must use ESPjoy interface from Antonio Villena.  [https://antoniovillena.com/product/espjoy/](url) . It simulate keyboard
keys press. You also need the green USB to PS/2 adapter.

Piligrim-77, 2024-04-25:
Hi. I join the question, is there an ESPjoy interface diagram somewhere? I don't have the opportunity to buy (I'm in another country), but I could assemble it myself on a breadboard.

Gods69, 2024-04-25:
[https://github.com/dacarsoft/DB9_2_Keyboard/tree/DB9_2_Keyboard_ESPectrum](url)
You can use a generic ATmega 32U4 boards like Arduino Pro Micro or Arduino Leonardo.
Program it with Arduino IDE using "Arduino Leoardo" as board target.
And see on the .ino source code the GPIO pins to connect to the DB9 socket.

Piligrim-77, 2024-04-25:
> [https://github.com/dacarsoft/DB9_2_Keyboard/tree/DB9_2_Keyboard_ESPectrum](url) You can use a generic ATmega 32U4 boards like Arduino Pro Micro or Arduino Leonardo. Program it with Arduino IDE using "Arduino Leonardo" as board target. And see on the .ino source code the GPIO pins to connect to the DB9 socket.

Ok, thank you.

Gods69, 2024-04-25:
Also You must have to use a AVR programmer because you have to program the ATmega chip by this ISP pins.

Gods69, 2024-04-25:
On the ATmega32U4 compatible card you use the USB socket (micro-usb ?) with a standard USB cable. On the another end you use a little green USB<->PS/2 adapter [https://share.temu.com/5SZx1jz1ZpA](url) to the PS/2 ttgo vga32 socket.
You must find +5V somewhere to power the 32U4 board.

soviet9922, 2024-04-25:
> On the ATmega32U4 compatible card you use the USB socket (micro-usb ?) with a standard USB cable. On the another end you use a little green USB<->PS/2 adapter [https://share.temu.com/5SZx1jz1ZpA](url) to the PS/2 ttgo vga32 socket. You must find +5V somewhere to power the 32U4 board.

Got it also, can't find a way to pass the verification on the image linked:  [https://share.temu.com/5SZx1jz1ZpA].
Can you upload it to anther image hosting ?.

Gods69, 2024-04-26:
OK. Good luck. Don't forget to swap (uncomment) the define lines on the source code :
#define AVILLENA_BOARD
//#define BLUEPILL_BOARD

Gods69, 2024-04-26:
ISP pinouts to program the chip. (you can use a Arduino UNO as AVR programmer)
[image]

soviet9922, 2024-05-01:
Having some issues compiling the 

C:\Users\soviet\Documents\Arduino\DB9_2_Keyboard\DB9_2_Keyboard.ino:86:2: error: 'KEY_UP_ARROW' was not declared in this scope
 {KEY_UP_ARROW, KEY_DOWN_ARROW, KEY_LEFT_ARROW, KEY_RIGHT_ARROW, KEY_RETURN, KEY_RIGHT_ALT, KEY_ESC, KEY_F12, 'z', 'y', 'x', 'm'},

Gods69, 2024-05-02:
I also have this issue. I think when i have changed to the right board type on the menu the error disappear.Sorry. I'm in Portugal, i can't help until my return.

EremusOne, 2024-05-12:
Thanks all for taking your time for finding and posting this information.

When I have the time, I wish to start a wiki and put together this info and info about many other things.

Regards! ;)

Gods69, 2024-05-12:
Program it with Arduino IDE using "Arduino Leonardo" as board target. No errors or use the .HEX.file:
[DB9_2_Keyboard_ESPectrum.zip](https://github.com/EremusOne/ESPectrum/files/15287382/DB9_2_Keyboard_ESPectrum.zip)

Gods69, 2024-05-17:
ESPjoy interface from Antonio Villena:
[image]
[image]

closed 2024-06-12 by EremusOne

soviet9922, 2024-08-10:
Here again, finally got the pro micro arduino and from the source code this is the pinout:
Bluepill ATmega 32U4 board pinout:
Arduino 2 -> DB9 pin 1: Up
Arduino 3 -> DB9 pin 2: Down
Arduino 4 -> DB9 pin 3: Left
Arduino 5 -> DB9 pin 4: Right
Arduino + -> DB9 pin 5: VCC 5V
Arduino 6 -> DB9 pin 6: A / Primary fire
Arduino 7 -> DB9 pin 7: Selection signal (allows more buttons)
Arduino - -> DB9 pin 8: GND
Arduino 8 -> DB9 pin 9: SELECT / Secondary fire

But there's nothing about the second controller, how can i connect it ?.

Gods69, 2024-08-11:
For the second DB9 joystick is:

Bluepill ATmega 32U4 board pinout:
Arduino 10 -> DB9 pin 1: Up
Arduino 16 -> DB9 pin 2: Down
Arduino 14 -> DB9 pin 3: Left
Arduino 15 -> DB9 pin 4: Right
Arduino + -> DB9 pin 5: VCC 5V
Arduino 18 -> DB9 pin 6: A / Primary fire
Arduino 19 -> DB9 pin 7: Selection signal (allows more buttons)
Arduino - -> DB9 pin 8: GND
Arduino 20 -> DB9 pin 9: SELECT / Secondary fire

EremusOne, 2024-08-11:
Hi!

Hash6iron, one of our users, has created a fork of ESPjoy firmware with all the info for building your own adapter. He called it "PowaJoy" :)

Here's the url: https://github.com/hash6iron/DB9_2_Keyboard_ESPectrum

Regards ;)

soviet9922, 2024-08-13:
It's me again xD
Asking things, now programed the arduino pro micro from the IDE and it works when i connect it to my pc using USB the  joysticks ports work great and get the keys output when moving them.

The issue is that using the usb ps/2 converter can't get it to do anything, the lights turn on the arduino. When i move the joystick but there's no output on screen.

Tested this in the vga32 but also nothing using ps/2 already enabled the second keyboard from the menu.
Tried several cables and adapters but seems not to be the issue.

Is like the arduino only works on USB mode.

Thanks !.

Gods69, 2024-08-13:
The arduino with its Atmega_328p only simulates a PS/2 keyboard. 
It does not have a USB mode, for that you have to use the ATmega32u4.
The small green USB/PS_2 converters do not have any electronics inside. 
The computer or keyboard must be able to work in PS/2 mode.