EremusOne/ESPectrum #52: Unable to open SD card

RA-NGR, 2024-06-07:
[image]
Module: TTGO VGA32 
ESPectrum version: 1.2
I made simple SD card test scketch
[image]

Gods69, 2024-06-07:
No problems here. Have you tried another brand of SD card?

EremusOne, 2024-06-07:
Hi @RA-NGR !

SD card is working flawlessly for everyone in all the boards tested with ESPectrum 1.2 so please give me some context about your issue:

- Exact model of Lilygo board (v1.2 or v1.4).
- How do you installed ESPectrum in the board: flashed with Espressif tool or built through repo code+VSCode + PlatformIO.
- Format of the card you are using.

Also, some video of the problem will be helpful.

Regards ;)

RA-NGR, 2024-06-08:
Hi @EremusOne 
The board is v1.4
ESPectrum.1.2.bin flashed with Espressif tool (I tried flash v1.1 - SD not working)
I trid both FAT and FAT32
Video is preparing, give me a couple of days

Hi @Gods69 
The card fully working with my test sketch (I take it from Arduino SD card example and slightly modify it). So problem isn't brand

Gods69, 2024-06-08:
I also have the TTGO VGA32 v1.4 with no problems. 
I'm using a old 1GB Sandisk SD card and updated to FW 1.2 via "SD card update" method.

RA-NGR, 2024-06-11:
Hi. Victor.
Here  http://www.atpilot.ru/IMG_4584.MOV is the video, but I think it is not helpful

Gods69, 2024-06-12:
Hi,
You need to try another memory card. There are several ways to access an SD card via software and your card (512MB) may not support the way it is accessed.

RA-NGR, 2024-06-12:
> Hi,
> 
> You need to try another memory card. There are several ways to access an SD card via software and your card (512MB) may not support the way it is accessed. (Also try to format it in FAT32)

Ok, I will try. But card also working good in my "spectrum" [image]

EremusOne, 2024-06-12:
Hi @RA-NGR 

There are potential incompatibilities between SD card readers and various brands or types of cards. In its current state, ESPectrum, on the Lilygo TTGO VGA32 v1.4, has proven to be very reliable, and most users who have encountered problems have resolved them by properly reformatting or changing the card.

In any case, I do not rule out the possibility that the malfunction could be corrected via software and that it might be possible to get that stubborn card to work. Unfortunately, I cannot dedicate time to this right now, and I would also need the card to test different settings.

So, my advice is to test with a different and, if possible, newer and of a known and reliable brand (I use Sandisk Ultra 32gb and 64gb with excellent results).

One question: ¿That sketch you posted first was executed on the same Lilygo where ESPectrum doesn't read your card or in different hardware?

Regards

RA-NGR, 2024-06-13:
Hi, Victor.
The sketch executed on the same board and it's really strangely.
So I opened issue.
Ok, I will try another card, thank you.

closed 2024-06-15 by EremusOne