EremusOne/ESPectrum #64: Support for SPI TFT screens

cgreening, 2024-06-29:
I'm happy to add this if someone can point me in the right direction/tell me if it's feasible.

I'd like to add support for an SPI TFT screen.

Jason2866, 2024-06-29:
The used library fabgl https://github.com/EremusOne/ESPectrum/tree/master/components/fabgl does support LC Displays already.

cgreening, 2024-06-29:
Are there any example configs for switching the display over to an LC display?

EremusOne, 2024-06-29:
Hi Chris,

I'm glad you're considering ESPectrum for your single-board ESP32 ZX Spectrum project. It'll be exciting to see ESPectrum running on it!

Adding TFT support to ESPectrum should not be too difficult. Although I have no experience with TFT screens and their operation, they have been widely used in many ESP32 projects. There are two main considerations: hardware and software.

Regarding hardware, since your board doesn't have a VGA output, I assume there are enough spare GPIOs and resources for a TFT screen (though I don't know the exact requirements for those screens).

For the emulator, adapting video functions to render to a different framebuffer or sending screen info directly to the TFT device should not be too difficult. However, I'm unsure if TFT screens have an internal framebuffer or need constant refreshing like VGA screens.

As @Jason2866 pointed out, the FabGL library seems to support some TFT screens, but I haven't tested it. ESPectrum doesn't use FabGL for video; it uses modified bitLuni's libraries. It does use the FabGL PS/2 component with some modifications to support two PS/2 devices simultaneously.

Currently, I'm focused on CPCESP (the next Amstrad emulation for the ESPectrum project) and have a roadmap to complete, so I don't have much free time to study it closely, or work on this matter.

However, if you can send me some technical information (schematics, pin info) and a prototype board, I can try to find some free time to work with you on getting ESPectrum running on it.

Take care!

cgreening, 2024-06-29:
Yes - no problem with the pins - my current prototype drives the screen no problem - it's just SPI + an extra couple of pins so you can get away with just 5 GPIO pins.

My current system reads the spectrum screen memory and sends out blocks of pixels using DMA. There's some simple detection of whether a block of pixels have any changes so it only updates modified portions of the screen.

The actual code for doing it is very short. So I really just need to switch off whatever VGA code is running and trigger my code at the end of each frame.

I don't plan on supporting any of the clever effects where the border is changed on different scanlines or attributes are modified on the fly.

So probably the main thing would be working out how to switch off the VGA output and also how to run drive the TFT screen - I don't know if the TFT_eSPI code works without the arduino framework, but I don't really use much of the library at all.

I should have the next prototype boards in a few weeks and can send over my current code if you want.

jgauchia, 2024-07-09:
> Hi Chris,
>
> I'm glad you're considering ESPectrum for your single-board ESP32 ZX Spectrum project. It'll be exciting to see ESPectrum running on it!
>
> Adding TFT support to ESPectrum should not be too [...]

Hi, If you want, I can try with TFT screens :P

Skudetto, 2024-08-26:
Hi everyone!

The task of adding TFT screen support to ESPectrum is definitely achievable, and there are some great examples of it being done successfully. The Phase project, which inspired the creators of ZX-ESPectrum, already includes versions for both VGA and TFT screens, using the Adafruit_ILI9341.h library. This could be a solid foundation for adding SPI TFT support to ESPectrum. You can check out the project here: https://github.com/retrogubbins/pase.

Additionally, there’s a more recent solution: https://github.com/ESPboy-edu/ESPboy_ZX48_LinKeFong/. This emulator handles graphics through the TFT_eSPI library and could also be very useful for implementing Spectrum emulation on a TFT.

These examples can serve as inspiration for improving ESPectrum. Best of luck, and I wish you success in your creative endeavors!

cgreening, 2024-09-01:
I've hit a bit of a stumbling block - I'm using an ESP32-S3 - and that's not supported in the version of the IDF that this project uses.

Looks like doing an upgrade is doable. But I might have a look at pulling the emulator out and reusing it instead.

closed 2024-09-30 by cgreening

Skudetto, 2024-09-30:
Hi everyone!
I see the issue has been closed, but I wanted to check if you were able to resolve the problem with TFT screen support on the ESP32? This seems like a very interesting topic and could help many others working with similar configurations. If you have any updates or solutions, it would be great if you could share them.

I hope the community can benefit from your experience!

Thank you for all your efforts, and good luck with further development!