EremusOne/ESPectrum #78: Support for composite (CVBS) output

joker07cz, 2025-02-27:
Is it possible to build ESpectrum with CVBS output with 15.625 kHz mode instead of VGA? Using just one pin for video signal instead of 8 (2 for each R/G/B, H sync, V sync)? I want to connect it to an old analog TV with composite (yellow cinch) input only. Of course with some signal adjustments outside the ESP32 board to conform to CVBS specifications (voltage levels, impedance...).

Thanks in advance.

Gods69, 2025-02-27:
No need for that. You can buy the cheap "VGA2AV" converter.

[image]

https://www.amazon.fr/Adaptateur-Composite-VGA2AV-Audio-convertisseur/dp/B07W5TRW5J

Jason2866, 2025-02-27:
A few resistors may work too. https://electronics.stackexchange.com/questions/403637/vga-to-composite-circuit-does-it-work

Gods69, 2025-02-27:
The resistors solution only give a black and white screen and the VGA vidéo must have 15Khz sync output. 
The "VGA2AV" do a 31Khz to 15Khz downscaling, in color.

EremusOne, 2025-03-28:
> Is it possible to build ESpectrum with CVBS output with 15.625 kHz mode instead of VGA? Using just one pin for (analog) video signal instead of 8 digital (2 for each R/G/B, H sync, V sync)? I want to connect it to an old analog TV with composite (yellow cinch) input only. Of course with some signal adjustments outside the ESP32 board to conform to CVBS specifications (voltage levels, impedance...).
> 
> Thanks in advance.

Hi!

In the short term a cheap adapter like that posted by God69 is your best bet.

There are some projects I've seen that maybe could accomplish what you're asking for but not a priority right now sorry.

Thx anyway for your interest in ESPectrum!

closed 2025-03-28 by EremusOne