EremusOne/ESPectrum #80: NES gamepad on FabGL VGA32

CityAceE, 2025-05-14:
The main request from users is support in the emulator for the joystick. However, there is nowhere to plug it in. The only option is to plug it into the mouse port via an expensive adapter. Meanwhile, there are free IO and 3.3V pins on the board. At the same time, modern inexpensive gamepads for NES work perfectly well powered from 3.3V. Such gamepads are directly connected to Raspberry Pi. Are there any obstacles to support NES gamepad connected to free IO pins?

[image]

EremusOne, 2025-05-14:
>Are there any obstacles to support >NES gamepad connected to free IO >pins?

Hi!

Unfortunately, there are no free IO pins in any of the boards supported.

That's why I decided to develop ESPectrum joystick system that supports two up-to eight buttons DB9 joysticks and allows to heavily customize them.

Regards ;)

CityAceE, 2025-05-15:
I looked at the schematic again and saw that most of those “spare” pins are already utilized for SD. That's too bad. And there was hope... Thanks for the reply!

Miguel-hrvs, 2025-05-15:
@CityAceE Our best bet is making a diy espjoy with this software: https://github.com/dacarsoft/DB9_2_Keyboard
It's basically two female db9 ports connected to a pro micro board and an ssd1306 oled screen.