EremusOne/ESPectrum #81: Esp-idf v5 migration

Miguel-hrvs, 2025-05-16:
I'm trying to see if not much changes are required to make the project compatible with the new esp-idf versions. I'll make a pr if I achieve it. For the moment, I'll post some notes about the progress if someone is also interested in doing this. Currently, even that I include esp_timer in fabui.cpp, platformIO gives me some errors about not finding it, this happened to esp_wifi in the v6.3.0, but updating the idf version fixed it. Don't know if this has to be fixed in a newer version or if there's any workaround for this.

For esp-idf v5: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/peripherals.html  
Components->fabgl->cmakelists->esp_adc  
Platformio.ini: platform = espressif@6.9.0  
Update i2s.cpp in src ESP32lib: https://github.com/bitluni/ESP32Lib/blob/master/src/I2S/DMABufferDescriptor.h

ESPectrum.cpp include: #include "esp_chip_info.h" https://esp32.com/viewtopic.php?t=31615&start=30

https://github.com/espressif/esp-idf/issues/9382

Change "esp_spi_flash" to "esp_flash"

In

```cpp
// Get chip information
```

Add:

```
uint32_t size_flash_chip;
esp_flash_get_size(NULL, &size_flash_chip);
```

Comment:

```cpp
//printf("silicon revision %d, ", chip_info.revision);
```

Change:

```cpp
        printf("%dMB %s flash\n", size_flash_chip / (1024 * 1024),
                (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");
```

In

```cpp
IRAM_ATTR void ESPectrum::audioTask
```

Comment:

```cpp
    //pac.tg_num             = TIMER_GROUP_0;
    //pac.timer_num          = TIMER_0;
```

OSDMain.cpp

https://community.platformio.org/t/include-errors-detected-please-update-your-includepath-squiggles-are-disabled-for-this-translation-unit/10882/2

Change #include "soc/rtc_wdt.h" to

```cpp
#include "rtc_wdt.h"
```

Change los dos includes de abajo to:

https://github.com/ayushsharma82/ElegantOTA/issues/196

https://forum.arduino.cc/t/esp32-chip-inquiries-no-longer-seem-to-work/1273364

```cpp
#include "esp_private/esp_int_wdt.h"
#include "esp_private/esp_task_wdt.h"
#include "esp_chip_info.h"
```

Add below:

https://stackoverflow.com/questions/2799789/what-c-library-do-i-need-to-get-this-program-to-compile

```cpp
#include <sys/stat.h>
#include <unistd.h>
#include <fcntl.h>    /* For O_RDWR */
```

Add:

```
uint32_t size_flash_chip;
esp_flash_get_size(NULL, &size_flash_chip);
```

Change:

```cpp
    textout = " Flash size    : " + to_string(size_flash_chip / (1024 * 1024)) + (chip_info.features & CHIP_FEATURE_EMB_FLASH ? "MB embedded" : "MB external") + "\n";
    VIDEO::vga.print(textout.c_str());
```

In fabui and fabutils.cpp:

```cpp
#include "esp_timer.h"
```

In fabui.cpp:

(some deepseek help)

Fix for incompatible types:

```cpp
void uiApp::killTimer(uiTimerHandle handle) {
  // First cast to TimerHandle_t
  TimerHandle_t timerHandle = static_cast<TimerHandle_t>(handle);
  
  // Get the event handler
  auto dest = static_cast<uiEvtHandler*>(pvTimerGetTimerID(timerHandle));
  
  // Create the association pair with consistent types
  m_timers.remove(uiTimerAssoc(dest, timerHandle));  // Use timerHandle instead of handle
  
  // Stop and delete the timer
  xTimerStop(timerHandle, portMAX_DELAY);
  xTimerDelete(timerHandle, portMAX_DELAY);
}
```

(some deepseek help)

Similar error in another function:

```cpp
void uiApp::suspendCaret(bool value) {
  if (m_caretTimer) {
    // First cast the uiTimerHandle to TimerHandle_t
    TimerHandle_t timerHandle = static_cast<TimerHandle_t>(m_caretTimer);
    
    if (value) {
      if (m_caretInvertState != -1) {
        xTimerStop(timerHandle, 0);
        blinkCaret(true); // force off
        m_caretInvertState = -1;
      }
    } else {
      if (m_caretInvertState == -1) {
        xTimerStart(timerHandle, 0);
        m_caretInvertState = 0;
        blinkCaret();
      }
    }
  }
}
```

In fabutils.cpp:

```
#include "hal/gpio_hal.h"
```

(some deepseek help)

Fixed function:

```cpp
bool FileBrowser::format(DriveType driveType, int drive) 
{
    // Initialize Task Watchdog with new API
    esp_task_wdt_config_t twdt_config = {
        .timeout_ms = 45000,  // 45 seconds in milliseconds
        .idle_core_mask = 0,  // Don't trigger on idle tasks
        .trigger_panic = false // Don't panic on timeout
    };
    esp_task_wdt_init(&twdt_config);

    if (driveType == DriveType::SDCard && s_SDCardMounted) {
        // unmount filesystem
        char drv[3] = {(char)('0' + drive), ':', 0};
        f_mount(0, drv, 0);

        void * buffer = malloc(FF_MAX_SS);
        if (!buffer)
            return false;

        // create partition
        DWORD plist[] = { 100, 0, 0, 0 };
        if (f_fdisk(drive, plist, buffer) != FR_OK) {
            free(buffer);
            return false;
        }

        // make filesystem
        MKFS_PARM mkfs_parm = {
            .fmt = FM_ANY,      // Format option
            .n_fat = 1,         // Number of FATs
            .align = 0,         // Alignment
            .n_root = 512,      // Number of root directory entries
            .au_size = 16*1024  // Cluster size (16KB as you intended)
        };
        
        if (f_mkfs(drv, &mkfs_parm, buffer, FF_MAX_SS) != FR_OK) {
            free(buffer);
            return false;
        }

        free(buffer);
        remountSDCard();
        return true;

    } else if (driveType == DriveType::SPIFFS && s_SPIFFSMounted) {
        bool r = (esp_spiffs_format(nullptr) == ESP_OK);
        remountSPIFFS();
        return r;
    }
    
    return false;
}
```

EremusOne, 2025-05-20:
Hi Miguel,

Good job!

I tried long time ago to compile with IDF > 4 and the main problem was with bitLuni's VGA code. Being that IDF 4 worked ok, the only update I did was using the most advanced version of IDF 4. Also, I read somewhere that IDF 5 could be slower or leave less SRAM available.

Anyway, I'll give a try to your findings and, if everything works ok, we'll move to IDF 5.

Tyvm for your time ;)

EremusOne, 2025-05-20:
I cloned your IDF5 test and tried.

First try I didn't get video but it was easy to solve changing order of parameters of rtc_clk_apll_coeff_set here:

	if (Config::esp32rev > 0)
	{
		// ESP32 chip revision > 0
		rtc_clk_apll_enable(true);
		rtc_clk_apll_coeff_set(vidmodes[mode][vmodeproperties::r1odiv], vidmodes[mode][vmodeproperties::r1sdm0], vidmodes[mode][vmodeproperties::r1sdm1], vidmodes[mode][vmodeproperties::r1sdm2]);
	}
	else
	{
		// ESP32 chip revision == 0
		rtc_clk_apll_enable(true);
		rtc_clk_apll_coeff_set(vidmodes[mode][vmodeproperties::r0odiv], 0, 0, vidmodes[mode][vmodeproperties::r0sdm2]);
	}

Once done this it started ok :)

SD card doesn't seem to work and timings in Standard VGA doesn't look ok but it's, anyway, a good start :)

closed 2025-08-11 by EremusOne